georgerobotics/cyw43-driver #1: License for the firmware

woachk, 2022-07-05:
Hello,

If we use a third-party Wi-Fi stack instead of this one on the RPi Pico W, is the firmware in this repository covered by that same license? Or does a different one apply for the firmware used standalone?

Thank you,

dpgeorge, 2022-07-05:
> If we use a third-party Wi-Fi stack instead of this one on the RPi Pico W, is the firmware in this repository covered by that same license?

Do you mean the binary blobs in the `firmware/` directory?  That you want to use `43439A0-7.95.49.00.combined` with another wifi stack/driver?

woachk, 2022-07-05:
> Do you mean the binary blobs in the firmware/ directory?

Yes.

> That you want to use 43439A0-7.95.49.00.combined with another wifi stack/driver?

Exactly.

kripton, 2022-07-06:
I'd also be interested because when packaging the pico-sdk for distributions including the driver, having multiple licenses involved doesn't make things easier :/

lurch, 2022-07-21:
Relevant info here: https://github.com/RPi-Distro/firmware-nonfree/commit/d9c88346cab86e23394ebf6cb6cb3069602d29f4