georgerobotics/cyw43-driver #2: Is it possible to port this driver and still comply with the licenses?

9names, 2022-07-06:
Hi Damien! There is some interest in porting this driver to Rust for the RP2040.
But there is also some concern around the wording of the licenses.
1st clause of LICENSE.RP:

> 1. The software can only be used and redistributed in conjunction with RP2040
   or any other semiconductor device produced by the Licensor.
   
If we put said source code on github, we would be redistributing the source but not with a hardware device from the Licensor, so it is not clear that we would be complying with this license. 
If we don't comply with that and we're relying on the regular LICENSE:

> 3. Any redistribution, use, or modification in source or binary form is done
   solely for personal benefit and not for any commercial purpose or for
   monetary gain.

If I provide source code to others, they would benefit and so I can't argue that it's solely for my personal benefit.
I think you could argue we would not be complying with this either.

And, same as  #1, we aren't sure what license the binary blobs are under and whether we are allowed to use or redistribute them.

Any clarification of these points would be greatly appreciated. Thanks!

dpgeorge, 2022-07-07:
This is a good question.  My understanding is that a translated piece of work (from one language to another) retains the licensing, and the spirit of the licensing in LICENSE.RP is to allow the source (in whatever language) to be used on RP2040 microcontrollers.  So it might be OK to port it to Rust and retain LICENSE.RP, but I will need to get back to you on that point.

Also for #1, advice will soon be given.

Rek-Malorm, 2023-05-06:
Hi @dpgeorge, Has there been an update on this?
It would be much appreciated to get a confirmation