georgerobotics/cyw43-driver #4: Evaluate compressing the wifi firmware

peterharperuk, 2022-07-07:
It looks like we could save ~85KB of flash memory if we compressed the wifi firmware binary. Evaluate if the code size, run time overhead is worth it.

dpgeorge, 2022-07-07:
It would be great if a compressed version of the firmware could be sent over SPI to the 43439, so it's faster to copy.  But I don't think that's possible, the base firmware on that chip doesn't support any compression.  So the benefit would just be smaller firmware (which is still a good benefit).

As for runtime overhead, I don't think it will be much.  The time for downloading the firmware is dominated by the SPI transfer.

kripton, 2022-07-07:
I think @peterharperuk meant to pre-compress the firmware files before including them into the build and then de-compressing them on-the-fly directly buffer-by-buffer before sending them via SPI to the cyw-chip.
Here are some compression numbers (file size in byte - file name):
```
225240  43439A0-7.95.49.00.combined
139024  43439A0-7.95.49.00.combined.br
140996  43439A0-7.95.49.00.combined.gz
164856  43439A0-7.95.49.00.combined.lz4
122820  43439A0-7.95.49.00.combined.xz
137444  43439A0-7.95.49.00.combined.zopfli
146782  43439A0-7.95.49.00.combined.zst
426094  4343WA1-7.45.98.50.combined
282192  4343WA1-7.45.98.50.combined.br
286309  4343WA1-7.45.98.50.combined.gz
330717  4343WA1-7.45.98.50.combined.lz4
249680  4343WA1-7.45.98.50.combined.xz
279263  4343WA1-7.45.98.50.combined.zopfli
304148  4343WA1-7.45.98.50.combined.zst
```
with `.br` being Brotli (https://github.com/google/brotli)
.gz being gzip -9
.lz4 being LZ4 -9
.xz being XZ (you don't want to decompress that on a RP2040, just for comparison)
.zopfli (https://github.com/google/zopfli) generates a data stream that can be decompressed using gzip/zlib/deflate but is smaller and much slower to compress
.zst being ZStandard

Most of the formats can be decompressed on the rp2040 (see https://github.com/kripton/rp2040-compressiontest), however I would personally chose zopfli for the offline/pre-compression and the existing gunzip to decompress on the rp2040 right before sending the file over SPI