georgerobotics/cyw43-driver #14: Allow user provided firmware and wifi_nvram_image

kilograham, 2022-07-17:
# w.r.t. firmware, there is a `storage_read_blocks` function however it is not hooked up. if we could provide a callback (possibly enabled by `#define` to allow the driver to determine the size of the firmware and iterate over the blocks; this would help uses cases such as
    * https://github.com/raspberrypi/pico-sdk/issues/928 (sharing the firmware between multiple binaries)
    * https://github.com/raspberrypi/pico-sdk/issues/909 (compressing the firmware)

# for completeness it seems like someone might want to tweak the wifi_nvram_image, which could be done by just allowing them to inject a #define.

Note right now both of these pieces of code are somewhat peculiar to the bus transport; idk if this should be done for both SPI and SDIO (I don't see why not though)

dpgeorge, 2022-07-19:
See #15 for the part (2), to make wifi_nvram configurable.

peterharperuk, 2022-09-01:
Has this been fixed?

dpgeorge, 2023-03-09:
In 82d2037f6af74d585ed1f9cec9f4e8166c4e84b0 the firmware blob was changed to a header file.  So now the config value `CYW43_CHIPSET_FIRMWARE_INCLUDE_FILE` can be used to set which firmware to include.

closed 2023-03-09 by dpgeorge