georgerobotics/cyw43-driver #16: Allow user to specify hostname

kilograham, 2022-07-19:
Currently `CYW43_HOST_NAME` can be specified at compile time, and is used here https://github.com/georgerobotics/cyw43-driver/blob/cd4075fc7d8c835431f5bed0e332f5a193990797/src/cyw43_lwip.c#L158

I'm not sure if you can call `netif_set_hostname` after this method returns and have it take effect (i'm guessing not?)

If not, it would probably nice to put an un-owned `const char *` in driver state, which if NULL uses `CYW43_HOST_NAME` (alternatively it can just default to that).

The main use case for this is actually more from MicroPython than C where I've seen people asking for it (in C it is easy enough to just override the #define).

In either case, there is also this line, the purpose of which I don't fully understand

https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/src/cyw43_lwip.c#L179

dpgeorge, 2022-07-20:
> I'm not sure if you can call `netif_set_hostname` after this method returns and have it take effect

I think you can, this function just sets an internal pointer in lwIP, which is then used whenever the hostname is needed.

For the MDNS case, that needs to be looked at in detail because I think the MDNS initialisation needs to be moved so that it's done after the IP address is assigned.

dpgeorge, 2022-09-09:
I think we should remove mDNS from this driver; see my comment https://github.com/georgerobotics/cyw43-driver/pull/30#issuecomment-1241608691

peterharperuk, 2022-09-09:
Yes - I can see the sense in that. Let me look into it and get back to you.

closed 2022-09-09 by peterharperuk