georgerobotics/cyw43-driver #17: Uses outdated lwIP API?

lurch, 2022-07-23:
Related to https://github.com/raspberrypi/pico-examples/issues/247

Whilst I wasn't able to reproduce the originally-reported problem in that issue, curiosity prompted me to do a bit of digging anyway... :wink: 

https://github.com/georgerobotics/cyw43-driver/blob/main/src/cyw43_lwip.c#L182 does
```
mdns_resp_add_netif(n, mdns_hostname, 60);
```
however the third parameter to `mdns_resp_add_netif` was removed from lwIP on [13th November 2018](https://github.com/lwip-tcpip/lwip/commit/62fb2fd749ba0bf5fed8c494e840075663484347). Looking at the dates of the [lwIP release-tags](https://github.com/lwip-tcpip/lwip/tags), I guess this means that there was a backwards-incompatible API change between the 2.1.1 and 2.1.2 versions of lwIP :confused:  And annoyingly, even though https://savannah.nongnu.org/projects/lwip mentions the 2.1.3 release of lwIP (which agrees with GitHub), http://www.nongnu.org/lwip/ shows the Doxygen for the 2.1.0 release, which uses the [old API](https://www.nongnu.org/lwip/2_1_x/group__mdns.html#gaa619ac8f46a4b4021195720f0355cbeb) :man_facepalming:

dpgeorge, 2022-07-25:
In MicroPython we use the `STABLE-2_1_2_RELEASE` of lwIP.  The signature of the MDNS function in that release is:
```c
err_t mdns_resp_add_netif(struct netif *netif, const char *hostname, u32_t dns_ttl);
```
So it looks correct for that version.

I guess the point is, we should update to `STABLE-2_1_3_RELEASE` :smile:

lurch, 2022-07-26:
> In MicroPython we use the STABLE-2_1_2_RELEASE of lwIP.

Well that's kinda awkward, as [pico-sdk](https://github.com/raspberrypi/pico-sdk/tree/master/lib) uses `lwip @ 239918c` which appears to be the [current HEAD](https://github.com/lwip-tcpip/lwip/tree/master) of lwIP `master`.
ping @kilograham and @peterharperuk 

Does the code in `cyw43-driver` need to be conditional on what version of lwIP it's being compiled against?

Looks like lwIP's `STABLE-2_1_2_RELEASE` is dated 22 Nov 2018, `STABLE-2_1_3_RELEASE` is dated 10 Nov 2021 and `master` is dated `9 May 2022`.

Oh! Looks like [STABLE-2_1_3_RELEASE](https://github.com/lwip-tcpip/lwip/blob/STABLE-2_1_3_RELEASE/src/apps/mdns/mdns.c) _also_ has the function-signature `mdns_resp_add_netif(struct netif *netif, const char *hostname, u32_t dns_ttl)`, so I was clearly wrong in my earlier complaint that lwIP were breaking APis in minor version releases, oops :man_facepalming:

dpgeorge, 2022-07-26:
> Does the code in `cyw43-driver` need to be conditional on what version of lwIP it's being compiled against?

Yes it probably does.  This driver can support various lwIP versions if needed, and to do that would use `#if` conditions.

But right now I think everything works with `STABLE-2_1_2_RELEASE` and current HEAD of lwIP master branch?

lurch, 2022-07-26:
> But right now I think everything works with STABLE-2_1_2_RELEASE and current HEAD of lwIP master branch?

Yes, I believe so (the other people I pinged will know more about this than me). But I guess https://github.com/raspberrypi/pico-examples/issues/247 implies that things might break if different preprocessor-definitions are used? :shrug:

dpgeorge, 2022-07-26:
See https://github.com/micropython/micropython/pull/8960 for update of MicroPython to 2.1.3.

lurch, 2022-09-08:
Yeah, apologies for any confusion my misunderstanding caused - **all** the `2.1.x` versions of lwIP are API-compatible; so it's only the `2.1.x` versions and `master` that have slight API-breakage under specific circumstances. `STABLE-2_1_3_RELEASE` (as used by MicroPython) is the latest tagged release of lwIP and dates from 10 Nov 2021 whereas lwIP `master` (as used by pico-sdk) currently dates from 9 May 2022.

I'm happy for this to be closed if everyone is happy with the current situation?

commit df2c2e2a (2022-12-09): src: Remove mDNS from cyw43-driver.

The lwip mDNS API has changed which can cause problems building it.
The suggestion is that this code should not be in the driver anyway,
so remove it.

Fixes #17
PR #39 (2022-12-09): src: Remove mDNS from cyw43-driver.
The lwip mDNS API has changed which can cause problems building it. The suggestion is that this code should not be in the driver anyway, so remove it.

Fixes #17
closed 2022-12-09 by dpgeorge