georgerobotics/cyw43-driver #18: IPv6 support (multicast and MAC address filtering)

oddstr13, 2022-08-03:
To be able to properly implement IPv6, a way to enable/disable filtering for specific multicast mac addresses is required.
A way to set the `allmulti` flag is a viable workaround, but is not currently available in the library.

https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/src/cyw43_ll.c#L1859

For more details on my journey to enable IPv6 in the Arduino core so far, see earlephilhower/arduino-pico#732.
In particular, the [`multicast.diff` spoiler in this comment contains my hack to enable the `allmulti` flag](https://github.com/earlephilhower/arduino-pico/pull/732#issuecomment-1202990025). The code is not of PR quality, but could serve as inspiration for what the bare minimum required work to enable IPv6 would be.

Ideally we'd unfilter specific MACs, but I didn't find any references to such functionality in this repo, and I haven't found any documentation for the firmware.

peterharperuk, 2022-08-24:
Nice work so far! There does appear to be a way to register multicast mac addresses. Presumably it would be enough to just set the list on initialisation? https://github.com/georgerobotics/cyw43-driver/blob/main/src/cyw43_ll.c#L1896

oddstr13, 2022-08-24:
That function does look promising!
We need to be able to add and remove addresses at any time, but this could be done with a read/write in low level, and implementing add/remove on top of that in the high level api.

Someone else should look into this if interested, I don't currently have time to work on this, and likely won't for quite some time. (Christmas? Next summer?)

peterharperuk, 2022-11-22:
This is relevant? https://github.com/georgerobotics/cyw43-driver/pull/24

PR #24 (2022-12-09): Implement a multicast filtering api and enable v6 for lwip
This is related to #19 should resolve #18.
Thanks to the good hint in from @peterharperuk  https://github.com/georgerobotics/cyw43-driver/issues/18#issuecomment-1225913653 I was able to use the mac address filtering. With these changes applied, ipv6 works perfectly for me.
And hints? Should I split up the 2 commits into to separate PRs?
closed 2022-12-09 by dpgeorge