georgerobotics/cyw43-driver #20: Monitor mode support

DrSchottky, 2022-08-07:
Hi,
I'm using this driver with the Pico SDK and I've seen that the code contains a disabled function used to turn on monitor mode.
I'd like to play with raw frames so I turned it on and made my custom handler to dump the frames, but I'm running into some problems:

- If I don't keep doing stuff like wifi scans I stop getting data
- After less than 1min it looks like the wifi chipset stops responding (`[CYW43] do_ioctl(2, 263, 82): timeout`)

Is that the reason why you commented out that code? Do you have a working Monitor mode implementation?

Thanks

psmitty7373, 2023-09-12:
@DrSchottky , did you ever find a solution to this?  I have the same issue.  Monitor mode will rx. about 16 packets then stop.

DrSchottky, 2023-09-12:
Yes somehow I solved it (long time ago, don't remember the details).
Anyway nexmon team is currently working an a driver/fw for this chipset

jpacora, 2023-10-10:
Hi @DrSchottky can you share a example of how to capture packets with the Pico SDK? I was able to turn on monitor mode based on the micropython code
https://github.com/micropython/micropython/blob/9f835df35e1628974437cdf87334667a9e2ce3c5/extmod/network_cyw43.c#L457