georgerobotics/cyw43-driver #27: Open (unsecured) APs are not supported

adamkaplan, 2022-08-29:
Referencing this MicroPython issue https://github.com/micropython/micropython/issues/9016#issuecomment-1219896195

Password-less AP support is disabled, [ostensibly by mistake](https://github.com/georgerobotics/cyw43-driver/blob/2ab6ca93f9cd044bc6f35c1403b1284e4161294a/src/cyw43_ll.c#L2018). However, even if that is fixed, the 802.11 security event processing doesn't appear to support APs with security disabled.

Clarke-Jools-36, 2022-10-26:
I have replicated this issue on MicroPython v1.19.1-88-g74e33e714, any updates on a fix?

gwo0d, 2022-11-23:
Likewise, I have replicated this.

gwo0d, 2022-11-23:
Submitted a patch for this here: #45

dpgeorge, 2022-12-19:
Fixed in commits 61432f360748949ad06564193a93d9fdb5afe9dd and 2cf328d9e41603405a037a29e081a7d30dd519e6 (it can now connect with CYW43_AUTH_OPEN).