georgerobotics/cyw43-driver #29: cyw43_hal_get_mac_ascii is used but never defined

DusterTheFirst, 2022-09-04:
cyw43_hal_get_mac_ascii is referenced in the following code but is never defined, causing pico-sdk programs compiled with `LWIP_MDNS_RESPONDER = 1` to fail to compile.

https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/src/cyw43_lwip.c#L176-L183

peterharperuk, 2022-09-05:
This exists in the MP port. Presumably we just need to borrow that implementation.
 #define cyw43_hal_get_mac_ascii         mp_hal_get_mac_ascii

DusterTheFirst, 2022-09-09:
> it'll be the responsibility of the application to call `mdns_resp_add_netif()` with a valid hostname

I think this is fair. It also allows for greater control of the mdns which is likely wanted by c application developers.

dpgeorge, 2022-09-09:
> It also allows for greater control of the mdns

Yes, eg if the application does not want mDNS.