georgerobotics/cyw43-driver #47: cyw43_wifi_leave does not appear to work

jepler, 2022-11-30:
In CircuitPython, we are implementing support for access point: https://github.com/adafruit/circuitpython/pull/7101/files

We expected that calling `cyw43_wifi_leave(&cyw43_state, CYW43_ITF_AP);` would stop the current access point and allow for creation of a new AP with different properties. However, it doesn't seem to actually stop the AP.

Possibly relevant, we found that to stop station mode we had to call `leave` on both ITF_STA and ITF_AP. Otherwise, wifi_link_status(ITF_STA) continues to report CYW43_LINK_JOIN.

I do not have a standalone example of either of these problems smaller than CircuitPython.

peterharperuk, 2022-12-01:
I might be seeing things, but it looks a bit like it's not sending the right command(s). 0x69 looks to be something else entirely?

#define CYW43_IOCTL_SET_DISASSOC        (0x69) // 105 in decimal

Something needs to call cyw43_wifi_set_up with up set to false, then send ioctl 0x76. We should probably have something in the pico-sdk / examples to test this.

eightycc, 2024-03-07:
Some re-work in Circuitpython and pull #107 resolves this problem.