georgerobotics/cyw43-driver #51: Minimum RP2040 clock speed

bruelltuete, 2022-12-14:
Is there a minimum clock speed the RP2040 must have for the driver to work?
If I run it at 48 MHz then the test pattern check at https://github.com/georgerobotics/cyw43-driver/blob/main/src/cyw43_ll.c#L1468 fails.
At 96 MHz it's fine.

This was a bit surprising because the RP2040 drives the clock signal, so I was expecting it to be mostly indifferent to what speed it runs at.

I wanted to use `cyw43_arch_gpio_put()` from pico_cyw43_arch and hit this test pattern fail.

bruelltuete, 2022-12-14:
Alternatively, what do the different values coming out of SPI_READ_TEST_REGISTER mean?

peterharperuk, 2022-12-14:
I don't think we've tested this. If you slow the clock down I think you will slow down the pio state machines which will slow down spi. I would have hoped spi would continue to work when you slow it down. I guess it doesn't?

The spi test register should only return the value it expects. If it sees something else it's not talking to the chip properly.

dpgeorge, 2022-12-19:
On tests with an STM32F7xx as the host MCU, I did find that there was a minimum host CPU frequency for which the CYW43 would work.  It was something around 48MHz.  I never investigated the reason for this.

rbryson74, 2023-04-27:
Just adding my experience here. 
I had the same issue on a PicoW. After setting the clock with set_sys_clock_48mhz() the test pattern check would fail.
Setting the SPI_PROGRAM_NAME to spi_gap0_sample1 resolved the issue.

peterharperuk, 2024-08-13:
I've now tested the driver running from xosc at 12MHz, you can speed up the pio to compensate. The pio lives in the pico sdk so this shouldn't be a driver issue.