georgerobotics/cyw43-driver #53: Q: which cyw43_delay_ms are time-critical

bruelltuete, 2022-12-21:
More a question... There are a bunch of `cyw43_delay_ms()` in the code, sometimes delaying by 50ms sometimes just 1ms in a loop, etc.
Which one of these is time-critical?
I would like to define `cyw43_delay_ms` as a task-switch for my homebrew OS. 50ms is half an eternity I'd like the CPU to do some other useful work. But I can't guarantee that it'll switch back exactly after 50ms (it's cooperative multi-tasking).

bruelltuete, 2022-12-21:
For example `cyw43_ll_wifi_on` has a few, see https://github.com/georgerobotics/cyw43-driver/blob/main/src/cyw43_ll.c#L1854, that start to add up.
And `cyw43_ensure_up()` as well, see https://github.com/georgerobotics/cyw43-driver/blob/main/src/cyw43_ctrl.c#L158.
Those are prime candidates for a task-switch.

But then there are many more `cyw43_delay_ms(1)`, ie only delay 1ms.

And also `cyw43_delay_us()`, https://github.com/georgerobotics/cyw43-driver/blob/main/src/cyw43_ll.c#L1877, but that looks like it could be anywhere between 0 and 150ms.

peterharperuk, 2022-12-21:
My guess is that the delays are not that critical. They probably are needed but I'd be very surprised if waiting longer caused a problem.

dpgeorge, 2022-12-22:
All those delays you link to are "at least", ie it must delay for at least that amount of time.  It shouldn't do any harm to delay longer than that.

Ultimately, it might be a good idea to provide a fully non-blocking API, so there are no delays at all...

bruelltuete, 2022-12-22:
thank you both!

closed 2022-12-22 by bruelltuete