georgerobotics/cyw43-driver #63: Help implementing CY43 derivative driver

soypat, 2023-02-21:
Hello peeps,
I'm currently developing with TinyGo and wanted to implement CY43439 support to make use of the on board wifi capabilities.

I've gotten [thus far](https://github.com/soypat/cyw43439), been able to read/write *something* via an SPI bitbang implementation. I'm honestly not sure what it is I'm getting back from the CY43. Would it be too much to ask to get some pointers on how to go about testing the SPI bitbang implementation? I can't poke around with a logic analyzer so I really feel like I'm in the dark here.

soypat, 2023-02-21:
So right now I think I've figured I've gotta get [these two](https://github.com/soypat/cyw43439/blob/main/cleanroom.go) functions working (Read 16 bit registers, swapped and non-swapped). Once I get those working I think I'll be able to figure my way on my own. Those functions use the SPI bitbang implementation written [here](https://github.com/soypat/cyw43439/blob/main/bbspi.go)

dpgeorge, 2023-02-21:
The original version of this driver was prototyped in (Micro)Python, using software (bit bang) SPI.  Once that prototype got to the stage where it could send a ping packet the code was converted to C, and became the driver in this repo.  It was a lot of trial and error to get it working.  Good luck with it!

soypat, 2023-02-21:
So it can be done! That's great to hear! Do you know anyone who worked on it I could ask for source code?

dpgeorge, 2023-02-21:
I worked on it, but I won't be able to make the source available, sorry about that.

soypat, 2023-02-21:
It's all good. Knowing it's possible to bitbang the CY is invaluable to me. :raised_hands:

closed 2023-02-21 by soypat