georgerobotics/cyw43-driver #71: Unclear visibility of headers

urbasus, 2023-03-22:
`cyw43.h` is one of the public headers with a presumably stable API. Are there others?

Listing the public headers in the top README.md should clarify the set.

Separating private and public headers into different directories is one way of clarifying visibility, but seeing as the project has milage maybe it's best not to move files around.

Marking internal headers with [doxygen special command \internal](https://www.doxygen.nl/manual/commands.html#cmdinternal) is yet another alternative.

dpgeorge, 2023-03-23:
> Are there others?

I think this is the status of each header file:
- cyw43.h - public
- cyw43_country.h - public
- cyw43_stats.h - public (but optional)
- cyw43_sdio.h - public, only needed when implementing SDIO bindings
- cyw43_spi.h - public, only needed when implementing SPI bindings
- cyw43_btbus.h - public, only needed when implementing Bluetooth BUS bindings
- cyw43_config.h - internal (configured via user-provided cyw43_configport.h) 
- cyw43_debug_pins.h - internal / debugging
- cyw43_internal.h - internal
- cyw43_ll.h - internal

> Marking internal headers with [doxygen special command \internal](https://www.doxygen.nl/manual/commands.html#cmdinternal) is yet another alternative.

That's a reasonable idea, I'd be open to doing that.

urbasus, 2023-03-23:
I've done some digging. This is the solution I propose:

- Add `\file` to all headers.
- For private headers, add `\internal` just above `\file`.

Now for some rationale. From doxygen `\file` documentation:

> The documentation ... will only be included in the output if the file they are in is documented as well or if [EXTRACT_ALL] is set to YES.

Because `\internal` supresses the `\file` command documentation generated with configuration `INTERNAL_DOCS = NO` (default) will be absent of documentation for that file. Absence of documentation is in my opinion same as private.

This should neatly cover reading both source and generated documentation.

dpgeorge, 2023-03-24:
Sounds good!

dpgeorge, 2023-03-31:
Done in 8b7ba0b6340489c4795a7d7425ab58b5255730d4

closed 2023-03-31 by dpgeorge