georgerobotics/cyw43-driver #82: ioctl from udp receive callback fails

thingsconnected, 2023-06-22:
We found out that a LED toggle in our UDP receive callback sometimes causes second-long stalls in our firmware. The LED is connected to a GPIO pin of the CYW43, and the LED is flashed briefly by toggling the LED pin on and off using `cyw43_arch_gpio_put(CYW43_WL_GPIO_LED_PIN, on);` that calls through to `cyw43_gpio_set()`. This is on an RP2040 with C code built using the pico SDK.

Normally, the LED toggle works fine, but sometimes the first ioctl fails and `cyw43_do_ioctl()` spins for a second before returning. The call to `cyw43_send_ioctl()` does succeed (as the code of `cyw43_do_ioctl()` assumes) but `cyw43_ll_sdpcm_poll_device()` only returns `ret == DATA_HEADER` and apparently the CYW43 forgot about the ioctl.

The following fixes both seem to work fine:
- after 10% of the timeout, the `cyw43_send_ioctl()` call is repeated once.
- after receiving a poll reply that is not related to the ioctl, the `cyw43_send_ioctl()` call is repeated once.

Both are a bit blunt and possibly risky as the code now may perform the same ioctl twice, which may not always be a good thing.

Does anyone understand what is happening here, and is there a proper way to fix it?

thingsconnected, 2023-06-23:
Seems like (part of) the answer is already in the question. Performing an ioctl may lead to processing of received data, so what happened is that the UDP receive callback was eventually called from within the UDP receive callback, with the ioctl calls stacking up (up to 4 levels deep in our case). This of course is Very Bad.

Not sure whether this is still an issue with the cyw43 driver then. Perhaps it would be wise make the driver foolproof by some way of locking and returning an error from ioctl, or by performing processing/callbacks from a queue rather than directly after receiving the data from `cyw43_ll_sdpcm_poll_device()`. I can imagine other scenarios where one would want to interact with the driver after receiving a packet, leading to comparable situations.

For now we will fix it on our end by not performing any significant work in the UDP receive callback, but setting some flags to be handled from our mainloop. Please advise whether there is a need to fix this in the driver in some way, in which case we can give it a shot.

peterharperuk, 2023-06-23:
It sounds a bit like you've found the problem and the issue is with your course. If you want more investigation an example that demonstrates the problem would be useful. But doing too much in callbacks is probably bad and deferring processing to your main loop is a good idea. On Pico you can use the async framework to defer work till later which in practice will be just after the driver has done its work.

thingsconnected, 2023-06-23:
Yes the issue was in our code, and fixed now, so no further investigation is needed. To me it is a bit surprising that blinking a led on UDP RX already causes strange issues.

I did notice this code and comments in `cyw43_ll.c`:

```
            int ret = cyw43_ll_sdpcm_poll_device(self, &res_len, &res_buf);
            if (ret != -1) {
                // CYW43_WARN("STALL(%u;%u-%u): got %d, %u\n", self->wlan_flow_control, self->wwd_sdpcm_packet_transmit_sequence_number, self->wwd_sdpcm_last_bus_data_credit, ret, res_len);
            }
            /*...*/
            if (ret == ASYNCEVENT_HEADER) {
                cyw43_cb_process_async_event(self, cyw43_ll_parse_async_event(res_len, res_buf));
            } else if (ret == DATA_HEADER) {
                // Don't process it due to possible reentrancy issues (eg sending another ETH as part of the reception)
                // printf("STALL: not processing ethernet packet\n");
                // cyw43_tcpip_process_ethernet(self, res_len, res_buf);
            } else if (ret >= 0) {
                // printf("cyw43_do_ioctl: got unexpected packet %d\n", ret);
            }
```

Here a reentrancy issue with DATA_HEADER responses from `cyw43_ll_sdpcm_poll_device()` is clearly identified and solved by dropping the packet. Why shouldn't the same logic apply to `cyw43_do_ioctl()`? I think my issue was exactly the kind of reentrancy issue the above comment is talking about.