georgerobotics/cyw43-driver #92: Tag v1.0.2 is not annotated

oliver-joos, 2023-10-10:
We use a Python script that calls `git submodule status lib/cyw43-driver` to retrieve the version and commit number of this driver. This relies on _annotated tags_ and failed until I added a note to tag v1.0.2 on my fork of this repo.
Please consider to annotate tag v1.0.2 in your repo as well.

dpgeorge, 2023-10-11:
Yes, annotating the tag is a good idea.  Previous tags have been annotated, and I forgot to do v1.0.2.

That's now fixed, and the tag has been force pushed to add the annotation.

To update a clone, use `git pull --tags --force`.

closed 2023-10-11 by dpgeorge

oliver-joos, 2023-10-11:
Thank you, now our versioning works!

Meanwhile I added my own tag annotation on my fork.
To overwrite the tags on my remote "origin" with those on your "georgerobotics" I did:
```
git fetch --tags --force georgerobotics
git tag --list -n99   # to check local tags and annotations
git push --tags --force origin
```

dpgeorge, 2023-10-12:
Great, thanks for confirming it works for you.