georgerobotics/cyw43-driver #102: Debug builds fail at both runtime and compile time.

iabdalkader, 2023-12-03:
Specifically on MicroPython's `mimxrt` port, when building with `DEBUG=1` at compile time the build fails because `storage_read_blocks` is not defined. I have to let the compiler optimize this function out by doing this

```diff
diff --git a/src/cyw43_ll.c b/src/cyw43_ll.c
index e2ac378..2077481 100644
--- a/src/cyw43_ll.c
+++ b/src/cyw43_ll.c
@@ -382,6 +382,7 @@ static int cyw43_read_backplane_mem(cyw43_int_t *self, uint32_t addr, uint32_t l
 static int cyw43_download_resource(cyw43_int_t *self, uint32_t addr, size_t raw_len, int from_storage, uintptr_t source) {
     // round up len to simplify download
     size_t len = (raw_len + 255) & ~255;
+    from_storage = 0;
```

And at runtime, debug builds assert because of the NVRAM overflow issue (which I tried to fix in #74 and was not merged) so I have to patch the files locally again to comment the asserts:

```diff
diff --git a/src/cyw43_ll.c b/src/cyw43_ll.c
index e2ac378..e1eb870 100644
--- a/src/cyw43_ll.c
+++ b/src/cyw43_ll.c
@@ -444,7 +444,7 @@ static int cyw43_download_resource(cyw43_int_t *self, uint32_t addr, size_t raw_
             sz = len - offset;
         }
         uint32_t dest_addr = addr + offset;
-        assert(((dest_addr & BACKPLANE_ADDR_MASK) + sz) <= (BACKPLANE_ADDR_MASK + 1));
+        //assert(((dest_addr & BACKPLANE_ADDR_MASK) + sz) <= (BACKPLANE_ADDR_MASK + 1));
         cyw43_set_backplane_window(self, dest_addr);
         const uint8_t *src;
         if (from_storage) {
@@ -477,7 +477,7 @@ static int cyw43_download_resource(cyw43_int_t *self, uint32_t addr, size_t raw_
             sz = len - offset;
         }
         uint32_t dest_addr = addr + offset;
-        assert(((dest_addr & BACKPLANE_ADDR_MASK) + sz) <= (BACKPLANE_ADDR_MASK + 1));
+        //assert(((dest_addr & BACKPLANE_ADDR_MASK) + sz) <= (BACKPLANE_ADDR_MASK + 1));
         cyw43_set_backplane_window(self, dest_addr);
         cyw43_read_bytes(self, BACKPLANE_FUNCTION, dest_addr & BACKPLANE_ADDR_MASK, sz, buf);
         const uint8_t *src;
```

It's just not practical to keep those local changes in every micropython fork I'm working on, or to add them every time I need to debug something, so can you please fix the debug builds somehow ?

dpgeorge, 2023-12-06:
See #103 for a fix to the first issue.

iabdalkader, 2023-12-06:
Thanks! I pulled that PR locally and tested, can confirm the debug build compile time issue is fixed, and WiFi is still functional.

dpgeorge, 2023-12-06:
> I pulled that PR locally and tested, can confirm the debug build compile time issue is fixed, and WiFi is still functional.

Thanks for testing.

dpgeorge, 2023-12-07:
The alignment issue should be fixed by #105.

dpgeorge, 2023-12-13:
Fixed.

closed 2023-12-13 by dpgeorge

iabdalkader, 2023-12-17:
I finally had a chance to test this, and everything seems to be working great. The debug build works as expected.. Thanks @dpgeorge for the excellent support!