georgerobotics/cyw43-driver #124: Compilation fails if LWIP_SINGLE_NETIF is set

peterharperuk, 2024-08-13:
If LWIP_SINGLE_NETIF is set netif_list is not defined. So you get a build error in the lwip code. Fix is simple enough.

```
diff --git a/src/cyw43_lwip.c b/src/cyw43_lwip.c
index be043f4..67d62b2 100644
--- a/src/cyw43_lwip.c
+++ b/src/cyw43_lwip.c
@@ -251,7 +251,8 @@ void cyw43_cb_tcpip_deinit(cyw43_t *self, int itf) {
         dhcp_server_deinit(&self->dhcp_server);
         #endif
     }
-    for (struct netif *netif = netif_list; netif != NULL; netif = netif->next) {
+    struct netif *netif;
+    NETIF_FOREACH(netif) {
         if (netif == n) {
             netif_remove(netif);
             #if LWIP_IPV4
```

dpgeorge, 2024-08-16:
Thanks for the report and fix.  See #125.

dpgeorge, 2024-10-07:
Fixed by 9d1008ce5804d5629c6a47ac7507d65524d38a59

closed 2024-10-07 by dpgeorge