georgerobotics/cyw43-driver #128: Infineon provides new FW version with vulnerability fixes

kolbma, 2024-12-01:
So not sure where this fw is coming from, but the versions seems to match the ones from Infineon.  
So I guess there should be also an update for the pico fw.

peterharperuk, 2024-12-02:
What vulnerability fixes are you referring to? Can you elaborate? The firmware comes from Infineon.

kolbma, 2024-12-02:
I can only guess from the fw change description in the linux driver release notes that it is related to deauth.  
https://www.infineon.com/dgdl/Infineon-Infineon_Linux_Wi-Fi_driver_FMAC_v2024_1115_release_notes-UserManual-v01_00-EN.pdf?fileId=8ac78c8c92bcf0b001934fb426884061

peterharperuk, 2024-12-02:
> from the fw change description in the linux driver release notes 

Can you please post a link to the release notes you're referring to? Thanks.

edmcman, 2024-12-28:
https://www.infineon.com/dgdl/Infineon-Infineon_Linux_Wi-Fi_driver_FMAC_v2024_1115_release_notes-UserManual-v01_00-EN.pdf?fileId=8ac78c8c92bcf0b001934fb426884061