georgerobotics/cyw43-driver #129: Async event 49 on Pico W

edmcman, 2024-12-28:
Hi,

I'm trying to troubleshoot a problem I'm experiencing in Micropython on a Pico W: https://github.com/micropython/micropython/issues/16482

The short description is that after long periods of inactivity, in certain power saving modes, the Pico W thinks it is connected but is effectively unreachable on the network.

When the problematic situation occurs, this message is repeated endlessly:

    [ 3505634] ASYNC(0000,49,0,0,0)
    [ 3507681] ASYNC(0000,49,0,0,0)
    [ 3508397] ASYNC(0000,49,0,0,0)

This message comes from this driver, which does not have a name for the 49 event.

I was looking at an another cyw43 driver, which lists event 49 as `ICV_ERROR`, suggesting that something has gone wrong with WEP/WPA.  Since the async event occurs endlessly and repeatedly, it suggests to me that the driver probably needs to respond in some way.  I'm at a disadvantage though because the cyw43 documentation does not seem public.

Is there a reason that this driver does not have event 49 defined or handled?  Is this something that can be remedied?  It seems like an event is occurring that needs to be responded to, but this driver simply ignores it.  I can't detect this situation from micropython, and the connection never recovers, so the situation is pretty frustrating!

dpgeorge, 2025-01-03:
> Is there a reason that this driver does not have event 49 defined or handled?

It's probably because this event was not encountered before, so no code written to handle it.

> Is this something that can be remedied?

Yes, we can definitely handle this event.  To start with, it would be good to have a way to reproduce this event reliably, to test the feature.

edmcman, 2025-01-03:
Hi @dpgeorge,

> It's probably because this event was not encountered before, so no code written to handle it.

That seems like a very good reason :-)

> Yes, we can definitely handle this event. 

Great!

> To start with, it would be good to have a way to reproduce this event reliably, to test the feature.

This is a bit tricky because it seems like the problem is triggered by a bug in my AP: [ActionTec WCB3000N](https://www.amazon.com/Actiontec-Dual-Band-Wireless-Extender-Ethernet/dp/B00FKTMWDE/ref=asc_df_B00FKTMWDE?mcid=966b83bafd2b33899160637806aa51b7&hvocijid=2924692306805255851-B00FKTMWDE-&hvexpln=73&tag=hyprod-20&linkCode=df0&hvadid=692875362841&hvpos=&hvnetw=g&hvrand=2924692306805255851&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9006783&hvtargid=pla-2281435177818&psc=1) Although it is a MoCa extender, you should not need a MoCa network to trigger the problematic WiFi functionality.

The bug (as I currently understand it) is that the Pico goes to sleep, a group rekey should occur, and the AP does not wake up the Pico to receive the group rekey.  I'm not sure how to replicate this... maybe in something like hostapd?  But I suspect the power saving stuff is all handled by the firmware.

Alternatively, I am happy to test changes.

edmcman, 2025-01-20:
Closed by c1075d4

Thanks @dpgeorge

closed 2025-01-20 by edmcman

dpgeorge, 2025-01-20:
Thanks for preparing the fix @edmcman !